KipK/openevse-gui-v2 #1: Missing library posix_tz_db ?

fhteagle, 2022-11-09:
I did a git pull into a fresh directory this morning (09 NOV 2022, ~1300UTC), ```npm install```, and ```npm dev run```. However, the GUI stalls with this error displayed onscreen and in npm terminal:

``` [plugin:vite:import-analysis] Failed to resolve import "../../../../library/posix_tz_db/zones.json" from "src/components/blocks/configuration/Time.svelte". Does the file exist?
44 |  import { utc2evseLocalTime } from "../../../lib/utils.js";
45 |  import { httpAPI } from '../../../lib/utils.js';
46 |  import timeZone from "../../../../library/posix_tz_db/zones.json";
   |                        ^
47 |  import { onMount } from "svelte";
48 |  import ButtonFetch from "../../ui/Button.svelte"; ```

Did npm not pull in all the dependencies?

KipK, 2022-11-09:
have you fetched submodules ? 
it's on /library/ folder

fhteagle, 2022-11-09:
Please excuse my being really, really rusty at using git/github.

``` git clone --recurse-submodules https://github.com/KipK/openevse-gui-v2.git ``` as the pull method fixed it.

Thanks.

closed 2022-11-09 by fhteagle

KipK, 2022-11-09:
you can 
git config --global submodule.recurse true

to set your git fetching automaticallyt submodules,