KipK/openevse-gui-v2 #5: Waiting and charging icons suggestions

fhteagle, 2022-11-10:
I am confused by the Charging status icon div using the class "is-warning" / yellow color? I would think this would be a normal condition, and green instead?

The waiting icon is a checked flag, which to me means "completed". Change it to an hourglass icon, or maybe a clock icon?

KipK, 2022-11-11:
I don't get it. First icon represent the Openevse state ( active / disabled ) 
The waiting Icons replace the Sleep Icon when state is active and no vehicule is plugued 

[image]

fhteagle, 2022-11-11:
If the OpenEVSE is active, the EV is connected, but EV is not actually charging (due to EV internal charging time configuration), the third icon is a checkered flag with green background. I will get my system into this state and take a screen shot later today.

fhteagle, 2022-11-11:
I updated the original post with screenshots for clarity about what I was talking about.

I think you changed the blue moon with tooltip "Sleeping" to be the blue thumbs up, correct? This is not what I was trying to indicate. Probably my mistake for not being clear enough.

What if the green background flag was changed to a blue background hourglass, with a tooltip "Waiting for EV to accept charge" ?

KipK, 2022-11-11:
oups nope I've made a mistake, now it(s the good ones

fhteagle, 2022-11-13:
Okay, the last icon being thumbs up and tooltip "Ready to Charge" for when the EV is not yet drawing power makes sense to me now. Thanks for fixing it!

Back to the charging icon being a yellow is-warning... I still think this should be green background for a normal condition.

KipK, 2022-11-13:
The idea here is to give emphasis to state "Charging" by introducing a unique color. 
Using same green as other states 
Choosed Yellow as it fits for a lighning icon ( perhaps using a lightning Icon is more suited in this case. 

Perhaps using Blue then ?

fhteagle, 2022-11-14:
Yellow bolt on green background makes visual and logical sense to me. Good compromise between "this is normal" and "careful, power is flowing". 

I wonder if a color blind person could see the yellow on green, though. I have a partial color shift, but not full color blindness. The white on green, or white on blue, or white on red each should have enough contrast to be legible. Yellow on green might be too close in contrast.

fhteagle, 2022-11-18:
Since you are using the "Pause" icon for OpenEVSE disabled in the status box, what about using that for Mode: OFF , instead of the circle and slash? 

Same suggestion about harmonizing Mode: ON icon and enabled status icon?

KipK, 2022-11-18:
I was about to change the pause icon too. Seems the circle & slah is more appropriated to a disabled state. 
Or perhaps we can change this icon depending we have an auto_release true or not . Pause Icon could be appripriated for an auto release true, the other one when false

New thing I've just added, tell me what do you think about this:
chrome-extension://mmeijimgabbpbgpdklnllpncmdofkcpn/app.html#/files/06c50c0d-916b-4977-y9c1-58717c51a754
It displays the current service that has the claim on the max_current setting here ( but could work like this for any property ) 
You can them from here delete any claim made by a service with lower priority than manual

KipK, 2022-11-18:
ok I'm testing this, not bad, and less Icon on screen for same message:
dizsabled:
[image]
active, waiting for EV
[image]
active, EV connected: 
[image]
active: charging:
charging:
[image]
error:
[image]

Then we have a bit more space on mobile to display a message

fhteagle, 2022-11-18:
Two icons is still very clear and yes a bit more simple!

As to the Screencast with the MQTT thing ... I had written up an idea to display the various amp targets from Eco, MQTT, etc, each on their own line. I think this might be better than trying to reuse one slider + manual/MQTT toggle....

KipK, 2022-11-19:
that's what is happening here. The slider will adjust the value to related claim that has the priority , and display the claim id as a tag under the slider. :)

There's no need to display the whole /claims topic for all the other claims that are not taken in consideration yet. The live one should be sufficient. 

I have iconified few things 
[image]

fhteagle, 2022-11-19:
Hmm. I have a different opinion on this. I think it would be useful to be able to see the action of each "amp controller",  and be able to easily toggle each on and off from the main page. Seeing a unified table of all the configured "controllers" is a great overview. And being able to see what would be happening with a particular controller, even if not active, would help the user to learn the system, confirm the system is configured as expected, and to notice unwanted quirks in what a controller is asking for. 

But I will admit that my opinion is not so well informed, as I have yet to configure any amp controllers on my home setup just yet.

KipK, 2022-11-19:
Imagine how bloated it would be as each property can have a different controller, so tracking the list off all the claims that are ignored for now.

Anyway, evse has no idea on who will have the priority but the main one.  Those fuzzy logic would have to run on client side.
Also those claims are always moving in real time,  priority change etc, .

The only thing we can display without ending with a blinking data christmas tree is the main order for each property.

closed 2022-12-09 by KipK