KipK/openevse-gui-v2 #6: 12 vs 24 hour clock consistency

fhteagle, 2022-11-10:
Much of the UI uses 24 hour clock (i.e. 7PM is 19:00), but some places use a 12 hour clock, such as the schedule input boxes.

Best of all possible worlds is to let the user pick 12 or 24 hour time format based on what they are comfortable with. Second best to be consistent with either 12 or 24 hour clock. I personally prefer 24 hour clock, but I know many people, especially in North America, who are not comfortable with this. 

Thoughts?

jeremypoulter, 2022-11-10:
This is also a problem for the LCD, could do with a config settings for 12/24 hour?

KipK, 2022-11-11:
The native time picker is handled by your OS settings , It displays 24h time format here.
But I can force AM/PM time on displayed time depending of the region. I already made something like this here, as it will display DD/MM/YY or MM/DD/YY depending of the region.

KipK, 2022-11-11:
Can you confirm me the time input box now has correct size to display am/pm ?

fhteagle, 2022-11-11:
Scheduled time input box is now wide enough to support AM/PM display, yes.

Can you let the browser use configured locality to display "Next Event" and the schedule time lines as 12h or 24h format?

KipK, 2022-11-11:
now all time input use system configured time format, so this should be unified on the UI ( but the current/next event for now as they display time format used by api for now) , can you confirm ?

fhteagle, 2022-11-11:
All inputs appear to be in browser controlled 12 / 24h in commit cccdb8d4934974b0c49597cc4345eb0cd4f15607 , yes. Some GUI display only elements are still forced to 24h time as you said.

KipK, 2022-11-11:
this should be adressed in last commit.

closed 2022-12-09 by KipK