KipK/openevse-gui-v2 #8: Status block sometimes stops updating

fhteagle, 2022-11-11:
I discovered a case where the status block stops updating. I do not have a way to solidly reproduce just yet, but I will keep trying. Might be just an in progress dev bug. Opening this just to remind myself to test more later.

Screenshot showing status bar has not updated in over 4 minutes:

[image]

KipK, 2022-11-11:
try to catch what logs has thrown, like web socket disconnection / error
Also go in Developper window / Network / get the /ws request, and check the message tab.
You'll see there if the websocket is still receiving data. If not I'd like to know why, as there's some reconenction logic in place.

fhteagle, 2022-11-12:
Found this in browser dev console, don't remember seeing it before:

[image]

KipK, 2022-11-12:
those are not related. There will be a lot of warning and stuff as it's a dev in progress. Don't bother on them.

the registerSW.js missing is normal it's for progressive webapps 

however some stuff won't works as expected from now, as it needs changes on the evse firmware that are not merged yet.

KipK, 2022-11-15:
this should be fixed now. Reopen if not.

fhteagle, 2022-11-15:
Test commit 15c159c2276e0fe5a7e747af1c21efd02d8cf118   against 7.1.3 / 4.1.5 physical OpenEVSE, but I am not seeing the clock in the upper right advancing to keep up with the current time. I assume this means status block is not updating. Is this because I am not running your dev branch firmware?

fhteagle, 2022-11-15:
Also, I appear to not have the privileges necessary to re-open this issue.

KipK, 2022-11-15:
The clock thing was a wip I have forgottent to finish :) Thanks 

It should be good, I don't think  you need the new build for that.

fhteagle, 2022-11-16:
At commit c7b3078fb93058547da433a6d3b6646a54ecd3f4

So the current time clock in the upper right advanced for about a minute, then froze again. Tried again and got the same results 3 different times in 3 different browsers.

Changed to Mode: ON in one browser, but the mode change did not get reflected on a different browser on my mobile device (still had the robot head timer lines, not the manual line). 

Amps, Power, Delivered, etc. seem to update as expected. Elapsed seems stuck though.

KipK, 2022-12-09:
should be ok, please ask for reopen if not.

closed 2022-12-09 by KipK