KipK/openevse-gui-v2 #9: FYI: Time handling bug affects Current Event and Next Event time display

fhteagle, 2022-11-11:
Looks like the time bug in https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/467 and https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/302 can affect the event display in status block. Opening this as a reminder to confirm fix after those upstream bugs are fixed.

KipK, 2022-11-11:
I've fixed this also time should be uniform now, thanks

fhteagle, 2022-11-12:
With commit 37ee7437c49b877ed87b15f5052aaf2a469046eb : 

Time display of Event and Schedule lines are showing erroneous times. Correct values are supposed to supposed to be disabled at 14:30 aka 2:30PM local and enabled at 23:00 aka 11PM local for the below screenshot (I checked /schedule endpoint and the text values are correct).

[image]

KipK, 2022-11-12:
I think I have fixed it now.

fhteagle, 2022-11-12:
At commit: 6854250b3fa094650959c3cd12bb6d0c17dd2bdd

Event lines are correct, but schedule table still has erroneous time:

[image]

KipK, 2022-11-13:
this is an outdated version, I think it was fixed while you've tested this one ;)

fhteagle, 2022-11-13:
Lol yeah why I am careful to reference the commit that git pull got me up to. 

Tested again at commit ending in 144e58 , but still seeing wrong time values in the schedule edit table. Same erroneous values for lines "2" and "3" as my previous screen shot.

KipK, 2022-11-13:
Mmhh that looks strange, as it use exactly the same function for displaying time from timers & plan.

Here it works : 
[image]

fhteagle, 2022-11-13:
A couple more data points:

I deleted my schedule table (and noticed the Auto button goes away with an empty schedule, which is a nice touch!). Re-added my schedule lines and checked on /schedule endpoint, and all seemed correct there. 

But I changed my time zone to NTP based Europe/Paris just to test. The /schedule endpoint raw values stayed the same, the event lines in the status block changed to the wrong values (same as schedule edit lines), and the schedule edit lines stayed the wrong time.

BTW, time input for schedule line items is back to being 24 hour based input, I thought it was allowed to follow the browser config?

KipK, 2022-11-13:
the time/date input fields in Browsers comes from OS regional settings.
The "toLocaleString" javascript fiunction use the browser regional settings.

That oculd explain you have a different display while testing changing just your browser language.

KipK, 2022-11-14:
I have changed for the Nuxon time lib yeserday, this cleans up lot oif time manipulation.
Is it still wrong ?

fhteagle, 2022-11-14:
at commit: 0dece329ce867a10133fe91298f0cf85cd7677b2

- Editing schedule table gives correct values in /schedule endpoint
- Status block event lines are correct (I like the word "timer", good change there)
- Schedule table lines are correct
- On Configuration > System page, Textbox is dd/mm/yyyy --:-- when NTP is selected, initially when Manual time is selected, and after a manual set time operation
- Manual set time operation works (checked against /status endpoint)
- Use Browser Time button does NOT work

Two steps forward, minor step back

fhteagle, 2022-11-15:
At 9f04ea5632dde4030e44a17d82beaa51b2c5bffe 

"Use Browser Time" button still makes the time input text box go to dd/mm/yyyy --:-- .

KipK, 2022-11-15:
yeah I have fixed that, but it's now on a branch I can't publish yet.
I'm rewriting the data management logic and from now everything is broken.

KipK, 2022-11-15:
Lof of fixes & improvment in latest commit. 
However, you'll need a special OpenEvse WiFi build of my "testing" branch as my changes are not in main yet:

https://github.com/KipK/ESP32_WiFi_V4.x/tree/testing

fhteagle, 2022-11-15:
I want to leave my OpenEVSE on 4.1.5 for at least this week, as I will be driving and charging quite a bit. I can test your firmware build after the 19 Nov probably.

"Use Browser Time" button appears to be fixed in commit 68654ca8891a2b27be8970e77063540bb25bb9c6

closed 2022-12-09 by KipK