KipK/openevse-gui-v2 #10: Remove ID from schedule table

fhteagle, 2022-11-15:
The left most column of the schedule table is schedule item ID. This does not have much meaning for the user. Important for the system, yes, but the ID is not something we can edit, or need to know, or affects function from our perspective. Table would be cleaner and clearer without this column.

KipK, 2022-11-15:
It's intended, to recognise without rolling over the schedule identical Timers with different days if you already know them.

fhteagle, 2022-11-15:
Hmm. Some ideas. Use one icon that means "this schedule is active every day" , and a different icon(s) for either A) a single one that means "this schedule ID is only active some days" , or B) seven checkboxes for the active days on the schedule table.

Better, though, I think, are the examples that JeremyPoulter gave of how smart thermostats show the schedule graphically, one row per day, left to right being from midnight to midnight showing on/off/heat/cooling /etc.  Graphical bars showing the schedule is the most slick, if you can pull it off.

KipK, 2022-11-18:
I've removed the ID for now. Will see later for the calendar range setting. It's tricky to display like this as we can have short timeranges to display.

closed 2022-11-18 by KipK