KipK/openevse-gui-v2 #12: Set Amps slider would not move

fhteagle, 2022-11-16:
At commit: 733fde87b8f44886cbef211c284ea67339ad6678

Set Amps was at 6A since several days previous. I could not move the slider to change the amps delivered to the car. Tried in ON, AUTO, OFF modes, but in each case the slider would refuse to change. 

Went back to 4.1.5 GUI and was able to change the Amps to 16 there. Reloaded new GUI and the 16 amps was then shown.

KipK, 2022-11-16:
please don't report bugs yet as I'm changing many thing, and there's bugs with /override on evse side I was not aware off.

KipK, 2022-11-16:
This is an active dev branch, only suggestions are welcome for now . We will have a test session when it's finished but it's mostly useless now. :)

KipK, 2022-11-17:
I think I've fixed all the bugs I got from firmware side, most of them where introduced by myself :))

It's testable now and adressed all the issues I had, but you first need a fresh build of my testing2 branch

closed 2022-11-17 by KipK