KipK/openevse-gui-v2 #13: About states colors

KipK, 2022-11-22:
On the old UI, I thought there was too much state color displayed around the status bar, make it confusing.
To keep it clear for users, I propose to keep only 3 color states:

- Disabled ( red ) 
- 
[image]

- Active ( not charging ) ( green ) 

[image]

- Charging ( yellow )
When charging, all bolts Icons on page will display also in Yellow.

[image]

Any thoughts on it ? 
@jeremypoulter
@fhteagle

jeremypoulter, 2022-11-22:
The colours around the status match the LCD/LED colours which is important for the US market

KipK, 2022-11-22:
Do we have to keep that on new UI? 

I mean, using blue for bolt icon , won't be really en phase

jeremypoulter, 2022-11-22:
I don't think we have to keep the colours in the UI, but there is a level of consistency with the LCD/LED (which are mandatory) if we do

fhteagle, 2022-11-26:
I really do not have a lot of preference about this. I personally rarely look at the physical display of the unit, and could not tell you what color means what on the display. So if even a very detail oriented person like me does not care, well......

Having the color meaning be consistent between the GUI and the display is nice on principle of course.

KipK, 2023-03-29:
@glynhudson @chris1howell could we consider changing the lcd led colors to more explicit ones?
I mean, Red -> Error is ok, but I'd rather have yellow -> charging, blue -> sleeping, green -> active and would match new UI better.

chris1howell, 2023-03-29:
LCD colors are limited.

We have 3 LEDs that can be ON or OFF

So we have
R0 G0 B0 backlight off
R1 G0 B0 RED (ERROR)
R1 G1 B0 PURPLE (SLEEP)
R1 G0 B1 YELLOW (Connected)
R0 G1 B0 GREEN (Ready)
R0 G1 B1 CYAN (Charging)
R0 G0 B1 BLUE (very dark not used)
R1 G1 B1 WHITE

I believe we have to use Green for ready, Red for error and blue for
charging but I will have to look at the specs.

KipK, 2023-04-10:
I've changed few things on state colors in UI to be consistent with leds.

Green, available, Yellow connected, fading from yellow to cyan blue while charging , red error states.

Also, @chris1howell I've put back the old logo for now. It's awful but let start with it for now. :)
That should be all good in latest gui_v2 daily build