KipK/openevse-gui-v2 #14: Half vs full width blocks, number of blocks per tab

fhteagle, 2022-11-26:
So I really like the icon bar across the bottom for the menu. Good call switching to that @KipK !

On desktop, though, the fact that some blocks are half the width of the status block (i.e.) schedule, and some are full width (i.e. charge) feels inconsistent to me. 

I do not think charge needs to be full width. Consider forcing it to half width and promoting the schedule block back to the default first loaded screen? Would cut down by one the number of icon bar items needed. 

I would be in favor of "always" showing two half width blocks per icon bar entry. Thoughts?

KipK, 2022-12-09:
It's just temporary layouts are not unified yet

KipK, 2022-12-09:
Schedule needs it's own page if we want to add an agenda like display later

closed 2022-12-09 by KipK