KipK/openevse-gui-v2 #22: Bottom Nav Bar issue when running Chrome on Android

aklnlh, 2023-01-13:
Just giving this an initial test and when using the new interface using Chrome (v: 109.0.5414.86) on Android (v:11) the bottom of what looks to be all pages are covered up by the bottom navigation bar.  At this time I've not tested any other browsers or OS.  I will do some more testing and update as I go. 

Is there any further information that you need to assist in resolving this anomaly.

KipK, 2023-01-13:
Can you post a screenshot here + resolution of your screen?

aklnlh, 2023-01-14:
The Phone is an Oppo A9 2020
Screen resolution is 1600x720

[image]

KipK, 2023-01-14:
it's normal it's called responsive design :) layout change according to available space. 
The page scroll downs if there's not enough space .
I've removed the scroll inside the box and now extend the box itself as it's more ergonomic on little screen

aklnlh, 2023-01-15:
I can slide the page down, but not enough to see the last two selections.  What you see in the screen shot is the page scrolled down as far as I can go, so I'm only able to view the top of the Developer box, but not the About box at all.

Once there is a new firmware to test, I'll install and test to let you know if you're changes have resolved this issue. Thanks

KipK, 2023-01-15:
Thanks I have reprroduced your issue.

KipK, 2023-01-15:
@aklnlh it's fixed now with many other things. You can grab new binaries
Have not finished yet to redo all the config forms. I have started to remove all unnecessary save button ( HTPP, MQTT for now )

closed 2023-01-16 by KipK

aklnlh, 2023-01-16:
I've updated and can confirm that the issue is no longer present. 
Thanks for your assistance.