KipK/openevse-gui-v2 #23: Feature Request : Daily kWh counter

TommySharpNZ, 2023-01-28:
Any thoughts on getting a daily counter added similar to the session and total counters?

KipK, 2023-01-28:
this is more a feature request for firmware .

If data is published through the API it can be easily added here.

TommySharpNZ, 2023-01-29:
Thanks @KipK I have logged a feature request (issue) on the main repository... Hopefully they pick it up...

fhteagle, 2023-01-30:
@TommySharpNZ  Not sure what use case is driving your request. However, I have been enjoying the utility of HomeAssistant logging and graphing all of my OpenEVSE use activity. Maybe that would get you the data / display you are looking for as well until this is implemented?

TommySharpNZ, 2023-02-15:
> @TommySharpNZ Not sure what use case is driving your request. However, I have been enjoying the utility of HomeAssistant logging and graphing all of my OpenEVSE use activity. Maybe that would get you the data / display you are looking for as well until this is implemented?

Would you mind sharing your config here on on the HomeAssistant forums?

I have tried to use the Total Usage sensor from OpenEVSE but I'm suspecting this only updates at the end of each charging session which means I get a chart with lots of jumps and not a proper usage line...
[image]

KipK, 2023-02-15:
Yes total usage is only incremented at the end of a session. 
What I did to counter that is feeding a virtual counter on smart home side, that use the total +session elapsed while charging. And just the total when session ended.

fhteagle, 2023-02-18:
@TommySharpNZ -

I was able to accomplish this by creating a helper of type "Integration - Riemann sum" using source sensor.openevse_current_power_usage . Then create an automation to reset the helper at the time of day that you wish.

pdhoogh, 2023-02-23:
Question: how/when does the OpenEVSE decide the charge session is a new session (reset to zero)? Now sometimes I want to start a new session with an energy limit and I need to add that to the current counter in order for it to work as I intend to, a bit weird...

I am not sure what the advantage would be to have a daily counter in OpenEVSE? I can see that in emoncms, but not in OpenEVSE functionality per se. Just my opinion.

KipK, 2023-02-23:
counter is reset at new vehicle connection.

fhteagle, 2023-03-06:
@TommySharpNZ  -

KipK's newest GUI revision has your feature, btw:

[image]

Not sure when it will make it to the nightlies, or to a full release, but it is on its way!

TommySharpNZ, 2023-03-06:
Nice Nice Nice!
Although if it's just in @KipK new UI does not mean it will not be available in the "api" that the HomeAssistant integration uses?

KipK, 2023-03-06:
It's on the api yes

TommySharpNZ, 2023-03-06:
This is great news, thanks team!

KipK, 2023-03-07:
Fixed in EnergyMeter fw branch, waiting for merge

closed 2023-03-07 by KipK