KipK/openevse-gui-v2 #27: FR: Simultaneous Active Limits

fhteagle, 2023-02-17:
Per RC3 issue thread, splitting this here:

It may be useful for some users to be able to set multiple limits, and pause on whichever is reached first.

pdhoogh, 2023-02-23:
Agreed. A practical case I would like to be able to do is charging in "Fast" mode with a system limit of e.g. 16 Amps and using the Shaper. I tried this in 4.1.5 but that did not work. Not sure it is possible now?

What happens if ECO mode is used and Shaper is still enabled?  That also did not seem to work well last year.

KipK, 2023-02-23:
I'm pretty sure you can't set the same mqtt topic for booth shaper & divert on current implementation. 
But it's out of topic here, @fhteagle talks about charge limit feature.

closed 2023-02-23 by KipK