KipK/openevse-gui-v2 #30: MQTT topic for ECO and shaper must be different?

pdhoogh, 2023-02-24:
This is v4.7.1 on GUI v2.

I tried current shaper and it does not seem to work. I put this on a different issue, but Guillaume thought it was out of topic, so I'm making a new one for this. I'm not sure if this is a GUI v2 issue or a 4.1.7 issue. I have no problem moving this to the "general/official" section.

[image]

Guillaume thought in the current implementation the MQTT topics for ECO and Shaper must be different. Since _Shaper_ was built separate as an add-on I understand why that is. In my opinion both should be using the same topic by default. as it is essentially the same "process value", only logical.

KipK, 2023-02-24:
you're still on the UI git repo ;)

Anyway, the way Divert is implemented, it needs one of the 2 mqtt divert feed to be filled to know wich divert type it uses. 
Using the same for Shaper will then cause issue and force Divert to Grid I/E. 
This would need first to have a proper setting for divert type.

pdhoogh, 2023-02-24:
OK, you want me to move this to the https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues ?

closed 2023-02-24 by pdhoogh

KipK, 2023-02-24:
Yes, also I have a fix already for that I'll PR today

KipK, 2023-02-25:
https://github.com/OpenEVSE/ESP32_WiFi_V4.x/commit/bab6d5a43f36aa0c58d02244945d4e9dd26c8ed6 should now allow using same topic, if you can test

pdhoogh, 2023-02-25:
> OpenEVSE/ESP32_WiFi_V4.x@ bab6d5a should now allow using same topic, if you can test

Thanks Guillaume! Will do tomorrow: drove 100km with EV today, SoC 19%, tomorrow lots of sun forecasted: ideal circumstances!

———————————
Philippe D’Hooghe
Meidoornlaan 10
9820 Merelbeke
+32 485 655 669

pdhoogh, 2023-02-28:
Bonjour Guillaume,

This specific item is working well now.

The behavior of 4.1.8 FW with the different modes: Divert Only, Shaper Only, Divert and Shaper both enabled I have also tested. Where would be the best place to put my test results with issues?

### Versions: 
- OpenEVSE HW is 2021 vintage
- OpenEVSE FW: 7.1.3.EU
-  WiFi FW reports 4.1.8. I also noticed 4.1.8 seems to have been pulled back out?

### The short story: 
- Shaper is not using minimum charge time. The ENABLE / DISABLE status changes in quick succession make the Peugeot e-208 charger decide things are wrong and latches it in the red "ERROR" mode: no more charging until you remove the charge plug physically from the charge port on the car and put it back in.
- With Divert & Shaper enabled, current is set by Divert and Shaper disables the minimum charge time. 
- Enabling / Disabling Divert and Shaper a few times confuses it, a restart of the WiFi is required.

The past few days I used the OpenEVSE with Divert enabled (Shaper Disabled, it needs work). That has worked well for me. There was lots of sunshine, an eventual cloud obscuring it, the swimspa (up to 5 kW power use) kicking in once in a while, my wife putting the washing machine and the dryer on, the dishwasher working, coffee makers spiking the power use... lots of opportunities for the OpenEVSE to go wrong. It did a wonderful job. The little Peugeot was happy and got charged from 19% to 75% SoC without issues.

By the way, Peugeot nailed it with the e-208. It is a fantastic little EV, ticking all the marks for comfort and utility for a city dweller. We use it for 95% of our errands. It is power hungry in winter (2020 model) and on the highway. We don't use it for trips that would need charging on the road for a one day stretch. Or if we need more room. For this we have the Merc Diesel, its usual spot is catching dust on the driveway.