KipK/openevse-gui-v2 #32: Cleanup: Energy Meter precision

fhteagle, 2023-03-06:
Session gives hundredths of a kWh (2 numbers right of decimal point), the rest of the items give only tenths (1 number right of decimal). It would be more legible to have all the same precision. I think tenths of a kWh is plenty of precision....

[image]

KipK, 2023-03-06:
We can display it with one digit here but I'd like to keep it 2 digits for the status bar and limits countdown. This gives a better feeling that energy is coming seeing the counter continuously increasing with elapsed counter.

fhteagle, 2023-03-07:
More precision on status bar makes sense to me too. Same precision in same "area" of the display is legible. Different precision, especially in columns, is harder to scan quickly.

KipK, 2023-03-07:
fixed in master

closed 2023-03-07 by KipK

KipK, 2023-03-07:
@fhteagle , I see in the screen capture above, your total is only 1.4kwh.
Is it a brand new openevse or have you reset the counter on the OpenEvse firmware ? 

Because it should import first time the old counter from OpenEvse. 
Would like to know if it failed or if its just because you had no data yet on it.

fhteagle, 2023-03-07:
@KipK -

Good catch. Lifetime on that unit was about 985kWh at the time that screen shot was taken. Looks like another display bug.

KipK, 2023-03-07:
@fhteagle total_energy was stored from the OpenEvse module, it can only be reset if you fully erase the Atmega before flashing it ( or perhaps with some RAPI commands but I don't think so ).
It is now handled from the wifi fw, so it seems it has failed on your side. 
Can you give me what $GU rapi command gives ? ( in Developper section )

fhteagle, 2023-03-07:
HA does not keep its own accumulation of hours, only what the OpenEVSE reports AFAIK.

> $GU
< $OK 7739264 990708^15

KipK, 2023-03-08:
The only total provided by the fw is total_energy, there's no other value shared.

KipK, 2023-03-08:
I've found the import issue, In fact it imported, but divided the kwh / 1000 ( thought it was watt ) . 

Here is a new build that fix it: 
https://drive.google.com/file/d/1FUoUDw5dxt1oNGdZbzbvlINi1uz9pAYp/view?usp=sharing

you can now reset the energyMeter by calling an http DELETE to /emeter with the following JSON in body:
{"hard": 1, "import": 1}

it should put everything to 0 then import the old counter

KipK, 2023-03-08:
it's not only the display, it's the data stored in the flash that where wrong.

That's why I'm surprised it wasn't updated on HA as it published the value without the import.  Probably HA missed the socket update