KipK/openevse-gui-v2 #34: No way in UI to prevent "Auto-Release"?

fhteagle, 2023-03-09:
Is there any way with GUI v2 to turn off / prevent / ban Auto-Release? I crawled through the whole UI and could not find it anywhere. I would like to be able to permanently turn off auto-release, in a way that will survive power cuts, firmware updates, etc.

KipK, 2023-03-09:
yes I had it disabled has it could never survive reboot & co so this has not much sense.

Would be better to add a default state in Configuration ( Default to Disabled | Active ) 
But that's a fw change

Would that deserve your use case ?

fhteagle, 2023-03-09:
The most important thing for my specific use case is the unit not changing itself on unplug for now. Since the auto-release switch is removed, is there any way that I can set auto-release to permanently off?

Long term, yeah that's a harder problem to capture all of the possible user desires for what the unit should do on power up, needs more discussion than just here. I will add that to the use case writeup project to try to plan/measure the population of users and what they want on that.

KipK, 2023-03-09:
auto release is never saved

fhteagle, 2023-03-09:
Having auto-release and/or default state of activated vs disabled be a user configurable item seems like better design to me as well. Let's get more input from the user base via https://github.com/OpenEVSE/ESP32_WiFi_V4.x/issues/461 on how to design it to make it work for the most users.

jeremypoulter, 2023-03-09:
But can you give some more info on your high level use case? It could be that there is an alternative way to meet your requirements. Manually adjustments on the UI are temporary by design

KipK, 2023-03-10:
I think just setting the default state to Active or Disabled in the config should handle this, or we could save manual overrides claims if set to auto_release=false ( will have to also enable auto_release for /override endpoint then  )

jeremypoulter, 2023-03-10:
I think it may be better to use the /claim API to set the active/disabled state, which you can then set the auto release to false. This then also frees up the override to be used as intended to temporarily override what Home Assistant has set.

KipK, 2023-03-10:
I guess we can save to flash the claims with auto_release = false, perhaps just for a new system client id with low priority to prevent flash writes.

fhteagle, 2023-03-11:
Agreed that claims is the "right" way for HA to be doing interaction with OpenEVSE. I will see what I can do to work the problem the other way around as well.