KipK/openevse-gui-v2 #36: Correctly display UI version?

glynhudson, 2023-03-13:
It's a minor issue, but I've noticed the UI version V1.0.0 is displayed. I think this should be V2.x.x when running the V2 UI to avoid any confusion with the V1 UI. 

[image]

KipK, 2023-03-13:
yeah mostly because I've named it differently and it was the first version of this one. but I join you, will be less confusing to directly start by a v2.0.0

KipK, 2023-03-13:
anyway I've removed the version number. It doesn't has much sense, it will be always delivered in bundle with the fw that is already versionned

closed 2023-03-13 by glynhudson