KipK/openevse-gui-v2 #37: Red colour should be reserved for error states?

glynhudson, 2023-03-29:
I've been in communication with @chris1howell and we both feel that red colour in the UI should be reserved for error states.

This is related to #13, I think the colours in the UI should match the colours on the RGB LCD where possible i.e sleeping / disabled is a purple colour. Although I'm not too fussed about the actual colour used, but I definitely think red should not be used unless there's an error. 

Currently red is used in a number of places in the UI 

[image]

chris1howell, 2023-03-29:
> Red is kind of of convention for stop/power off/disabling/Cancel...
>
> Error state are also marked in red as they are supposed to have switched
> the evse to a disabled state too.
>
> Using purple for such buttons, disabled status tags doesn't make much
> sense in a UX perspective.
>
> What could be an other way?

There is a lot of information on the subject, especially if you look at
SCADA systems. This is a great resource linked below.

Alarm color should be bright and not used for anything else. Grey may be a
good choose for disable.

https://realpars.com/hmi-colors/

KipK, 2023-03-29:
we are talking about UX conventions for main users not sure industrial conventions are really suited here

KipK, 2023-03-29:
@chris1howell 
I've removed the red color for everything than errors. It's a bit less appealing but this will fit for now. 
I'll have to introduce a new color in the palette somewhere else, dunno yet.
It's pushed on master, new builds compiling now. 

[image]

glynhudson, 2023-03-29:
Thanks, I've just flashed the new build. That's great, I like it. I think it looks more elegant, but importantly it will clearly highlight when there is an error. Having clear error reporting is very important for an EVSE.

closed 2023-03-29 by glynhudson

KipK, 2023-03-30:
I've added some visibility to errors, now monitoring nav button goes red if it gets arror and default jump to Security tab : 

[image]