KipK/openevse-gui-v2 #38: Toggle instead of button for Fast/Eco?

glynhudson, 2023-03-29:
What are your though on using a toggle switch to change between Fast and Eco? Currently, the user has to tap the button labelled "Fast" to enable "Eco", this seems a bit confusing to me and has caught me out a couple of times! I'm sure I'll get used to it overtime, but I think a toggle would be more intuitive? A toggle would also match the UI element above. What do you think? 

[image]

KipK, 2023-03-29:
Yeah it's a bit confusing on mobile. On desktop there's the tooltip when rolling over the button mentioning "switch to Eco".
Those are toggle buttons but I can probably improve this. 
Will check this

KipK, 2023-03-29:
[image]

Done by https://github.com/KipK/openevse-gui-v2/commit/8c6bfee1395241821792158e426c5751e9b83142

glynhudson, 2023-03-29:
Thanks a lot, I think that will be clearer. I think mobile is the primary way most people will interact with the EVSE.

closed 2023-03-29 by glynhudson

glynhudson, 2023-03-29:
Has the build with these changes in been pushed to master? I've just flashed the latest build (https://github.com/OpenEVSE/ESP32_WiFi_V4.x/actions/runs/4558985925#artifacts) and it does not have the Eco/Fast toggle