KipK/openevse-gui-v2 #40: Firmware update offers lower than installed version

fhteagle, 2023-03-30:


KipK, 2023-03-30:
4.1.9 is prerelease

fhteagle, 2023-03-30:
Okay, it is getting a little hard for the testing public to follow what branch/version/from what repo/google drive drop should be tested when. Seems like others are having trouble as well, as indicated by the tail end of this thread: https://community.openenergymonitor.org/t/openevse-new-user-interface-testers-translations-needed/22371/

Can we rethink the strategy of having the wifi firmware and GUI codebases in separate repos? Seems like they are not staying sync'ed and one is always ahead in some ways and behind the other in some ways....

KipK, 2023-03-31:
This because some PR takes time to be merged and so fixes are sometime just on my personal builds for testing . 

This will change after next major release when the UI2 will be considered as official. We are still in a kind of transition , and as you followed there was lot of stuff to fix on different fronts

fhteagle, 2023-05-27:
So to take this one step further:

I have the OTA updated dev build master_764b3174_modified on one of my units. If I go to  Config > Firmware > Update > Check "Dev", I see "V2 GUI pre-release
2023-05-10T05:28:39Z" is current. There is nothing in the version string of currently installed, or the string of available OTA that tells me if what I have is current or not. Therefore, I have to try to remember what I last installed and if it matches or is older than the available dev OTA.

Possible solutions:

1. Mod the build string to include a date string, such as master_764b3174_202305100528
2. Change the available OTA display to include the commit id instead of "V2 GUI ..."
3. Show the installed firmware's release datetime value on the update dialog?

Not sure what is easiest based on the github tags / json manifest / etc that @KipK and @jeremypoulter cooked up. Just looking for something I do not have to remember to be able to determine if I have the latest dev release or not.

I do not think greying out or disabling the Update button is a good solution, as it might be useful to reinstall the current OTA in case of a bad flash event?

KipK, 2023-05-27:
this has changed since the 4.2.2 release, it was in development stage. There's no v2 gui anymore, gui has been merged now.
Dev builds will still keep github short hash as version number