KipK/openevse-gui-v2 #41: add some check before to prevent OTA on _dev debug builds

KipK, 2023-03-31:
https://community.openenergymonitor.org/t/openevse-new-user-interface-testers-translations-needed/22371/250?u=guillaume_s

KipK, 2023-04-01:
Why not. What's GitHub is giving through the api seems enough, just need to fill one unused field with the release number/hash. 
Doing it with our own Json would probably simplify the thing.

KipK, 2023-04-02:
@jeremypoulter 
something like this generated for each build could be good enough:

```
{
	tag: "v2_gui" 
	name: "V2 GUI pre-release"
	version: "latest-1-g1fcd5d8"
	type: "daily" //( release , pre-release, daily )
	date: "2023-04-01T23:39:38Z",
	files: [
		{
		build: "openevse_wifi_v1"},
		url: "http://..../openevse_wifi_v1.bin"
		}
		
	]
}
```

```
{
	tag: "v4.2.0" 
	name: "V.4.2.0"
	version: "v4.2.0"
	type: "prerelease" //( release , prerelease, daily )
	date: "2023-04-01T23:39:38Z",
	files: [
		{
		build: "openevse_wifi_v1"},
		url: "http://..../openevse_wifi_v1.bin"
		}
		
	]
}
```

KipK, 2023-04-02:
But if you find a way to add the version for all types directly on github api /release feed, then we don't need external file.

( the get the "latest-1-g1fcd5d8" for the dailies, for now grabing the tag_name or name gives "v2_gui" or "V2 GUI pre-release" )

[image]

jeremypoulter, 2023-04-02:
Will take a look, might not be able to do the release/pre-release but will see. FYI I have changed the version format on the latest master as git describe is not consistent

KipK, 2023-04-02:
For release / prerelase the tag_name field is ok, for daylies we'll have to use something else