KipK/openevse-gui-v2 #42: Add the ability to set the time from the browser

jeremypoulter, 2023-04-01:
The V2 GUI is missing the ability to set the time on the host browser. This is really handy should you be in a standalone setup with no internel access. Obviously you can use the manual time but this needs you to carefully set the correct time on the UI first, in the vast majority of cases the browser will have the correct time so it would be great to just use that. Probably should also set the timezone of the browser as well (that being said NTP should also default to the time zone of the browser if it doesn't get a sensible one from the EVSE).

KipK, 2023-04-01:
when you set manual mode it automatically set to local browser time first. then you can change it if you want.

jeremypoulter, 2023-04-02:
This is likely to still be wrong though based on the time between switching to Manual mode and clicking on the button. In the v1 UI the time is read after clicking the button just before sending the request to the browser.

KipK, 2023-04-02:
it set the time when you set to manual with the selector, there's no need on clicking on the set time button again.
Doesn't it works as it should ?

KipK, 2023-04-02:
seems the get timezone from browser was not working as it should. I've fixed it on the shaper branch, will merge it later.

KipK, 2023-04-02:
spotted something, when you're already in manual time mode, then the only way to update to browser time was switching back&forth from ntp. 

I've added a button also

jeremypoulter, 2023-04-02:
Ah I see, not too sure that will work, as if you consider the off line case. For example consider an OpenEVSE used as a portable charger that is taken on roadtrips so might need to set the time every stop, but you would only switch to manual when you first get the unit.

KipK, 2023-04-02:
should work with the new button now.
Offline evse should definitevely have RTC, I can't imagine having to set the time each time you have to use it. :)

closed 2023-04-10 by KipK