KipK/openevse-gui-v2 #45: Make error state more prominant

glynhudson, 2023-04-04:
Currently, when the unit enters and error state on UI V2 it's quite subtle, and it's not possible to tell at a glance what the error is:

[image]

Compared to normal charging:

[image]

I think as a minimum the status border colour should glow red.

Personally I quite like the text on the old UI e.g Charging | Waiting | Vent error | GND fault etc. I understand if for UI aesthetics you prefer not to have text for 'Charging' or 'Waiting', however in the event of an error I would prefer the error message to be displayed clearly in text without having to mouseover or tap a logo.

KipK, 2023-04-04:
the error are detailed in the tooltip of the error icon. Roll over, or tap the icon to display it.
You have more details of the errors then in Monitoring section.

What error was thatnot disabling the evse ? It should be in red in error state

glynhudson, 2023-04-05:
That was a diode error, I also tested with a vent error, both disabled the EVSE, but the background colour did not go red

KipK, 2023-04-05:
[image]
Doesn't seems to be disabled on your screen ( I have to check that)

KipK, 2023-04-05:
Seems I have found why, fxed

closed 2023-04-05 by KipK

KipK, 2023-04-10:
I've added error message displayed in an alertbox and in monitoring/safety section.